jarzebski/Arduino-DS3231 #12: Day of the Year is incorrect

LukCaha, 2017-01-01:
Hi, I testing your library but Day of the Year is incorrect. I try print this dates with DotY:
```
2016-01-01, 0
2016-01-31, 30
2016-02-01, 32 (should be 31)
2016-02-27, 58 (57)
2016-02-28, 59 (58)
2016-02-29, 60 (59)
2016-03-01, 59 (60)
2016-03-02, 60 (61)
2016-03-03, 61 (62)
```
Same will be every leap-year.
Please, can you fix it?

novakalex, 2018-04-03:
In function **date2days** move **isLeapYear** to FOR.
```
uint16_t DS3231::date2days(uint16_t year, uint8_t month, uint8_t day)
{
    year = year - 2000;

    uint16_t days16 = day;

    for (uint8_t i = 1; i < month; ++i)
    {
        days16 += pgm_read_byte(daysArray + i - 1);
        if ((month == i) && isLeapYear(year))
          ++days16;
    }

    return days16 + 365 * year + (year + 3) / 4 - 1;
}
```

jarzebski, 2023-05-18:
will be fixed in 1.1.0 release

closed 2023-05-18 by jarzebski