jarzebski/Arduino-DS3231 #15: Alarm 2 settings needs to be corrected, no seconds

Eheran1, 2017-09-23:
This is out of the "alarm" example and will not work, see below for the working example. It took me way too long to figure that one out: There can not be any seconds with Alarm2.
```
  // Set Alarm - Every full minute.
  // DS3231_EVERY_MINUTE is available only on Alarm2.
  // setAlarm2(Date or Day, Hour, Minute, Second, Mode, Armed = true)
   clock.setAlarm2(0, 0, 0, 0, DS3231_EVERY_MINUTE);
```
```
  // Set Alarm - Every full minute.
  // DS3231_EVERY_MINUTE is available only on Alarm2.
  // setAlarm2(Date or Day, Hour, Minute, Mode, Armed = true)
   clock.setAlarm2(0, 0, 0, DS3231_EVERY_MINUTE);
```

rjaraman21, 2017-10-20:
I'm Sorry that I can't anything to this post because I'm new at this and having ALOT of trouble finding simple examples or anyone knowing how to set a simple alarm.
Everyone can tell me how to set the ds3231 clock, but not how to make a pin go high at 3;30 every Wednesday. Very frustrating. I have a watch for telling time, Hope the ds3231 can do more than that.

Eheran1, 2017-10-20:
This example really did not help you?
```
// Set Alarm - 10h:45m:30s in every Friday (1 - Mon, 7 - Sun)
// setAlarm1(Date or Day, Hour, Minute, Second, Mode, Armed = true)
// clock.setAlarm1(5, 10, 40, 30, DS3231_MATCH_DY_H_M_S);
```