jarzebski/Arduino-DS3231 #22: clock.dateFormat() doesn't output anything in Arduino 1.8.7

TomTomcrafter001, 2018-10-22:
After upgrading the Arduino IDE to 1.8.7 I only get blank output from clock.dateFormat(), even in the example sketch. Raw-output is still working (sketch DS3231_simple).

Once downgraded to 1.8.5 it's now working again.

1.8.7 complained about something in DS3231.cpp:
`DS3231.cpp:1064:16: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]`

TomTomcrafter001, 2018-11-17:
Add: Happens on 1.8.5 too, if AVR Core is newer than 1.6.20

erumaithi, 2018-11-27:
i am having the same issue ,  it seem the author have abandoned the library. this library have not been updated in 4 years.  it is a shame as it the best and the easiest.

tomjschwanke, 2018-11-27:
> 
> 
> i am having the same issue , it seem the author have abandoned the library. this library have not been updated in 4 years. it is a shame as it the best and the easiest.

Yep, probably gonna start looking for a replacement soon, but I'll have to recode lots of my sketches *sigh*

Ikar4n, 2018-12-02:
It was a simple fix.

[Arduino-DS3231-master.zip](https://github.com/jarzebski/Arduino-DS3231/files/2637387/Arduino-DS3231-master.zip)

erumaithi, 2018-12-03:
thank you that is amaizing

peiperakos, 2019-04-13:
Thanks its Works You save my day :)

Jeins98, 2019-06-01:
With the newest Arduino IDE it didn't work. You have to downlaod Arduino IDE 1.8.5. It worked like a charm. For everyone who doesn't get it working.

rwl7532, 2019-06-22:
> With the newest Arduino IDE it didn't work. You have to downlaod Arduino IDE 1.8.5. It worked like a charm. For everyone who doesn't get it working.

A few comments back is an updated zip file. It works with IDE 1.8.9

FesSpinosa, 2019-07-03:
For me, the library works perfect on IDE 1.8.9(Windows Store 1.8.21.0). Where I can find a list which return values are possible? Many thanks.

Jeins98, 2019-07-03:
> > With the newest Arduino IDE it didn't work. You have to downlaod Arduino IDE 1.8.5. It worked like a charm. For everyone who doesn't get it working.
> 
> A few comments back is an updated zip file. It works with IDE 1.8.9

Thanks, I will try it.

FesSpinosa, 2019-07-04:
Got the list!!

iBrick, 2019-10-29:
> 1.8.7 complained about something in DS3231.cpp:
> `DS3231.cpp:1064:16: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]`

Regarding this warning you may use quick fix in DS3231.cpp file: 
change lines:
char *daysLong[]  = {"Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday", "Sunday"};
char *daysShort[] = {"Mon", "Tue", "Wed", "Thu", "Fri", "Sat", "Sun"};
char *monthLong[]  = {"January", "February", "March", "April", "May", "June", "July", "August", "September", "October", "November
char *monthShort[] = {"Jan", "Feb", "Mar", "Apr", "May", "Jun", "Jul", "Aug", "Sep", "Oct", "Nov", "Dec"};

to 
const char *daysLong[]  = {"Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday", "Sunday"};
const char *daysShort[] = {"Mon", "Tue", "Wed", "Thu", "Fri", "Sat", "Sun"};
const char *monthLong[]  = {"January", "February", "March", "April", "May", "June", "July", "August", "September", "October", "November", "December"};
	const char *monthShort[] = {"Jan", "Feb", "Mar", "Apr", "May", "Jun", "Jul", "Aug", "Sep", "Oct", "Nov", "Dec"};

PeterPerry0, 2022-03-30:
> It was a simple fix.
> 
> [Arduino-DS3231-master.zip](https://github.com/jarzebski/Arduino-DS3231/files/2637387/Arduino-DS3231-master.zip)

hello sir, you can try maintaining this lovely library by hosting it and giving credit to the author, since he has stopped working on it

jarzebski, 2023-05-18:
fill be fixed in 1.1.0 release

closed 2023-05-18 by jarzebski